fivdi/epoll #12: Module did not self-register

ThumbGen, 2015-04-04:
Trying to use the onoff library inside my project but I always get a "Module did not self-register" error from the epoll module (bindings.js line 83).

Using latest nodejs version (tried with several other older versions back to 0.10.x).
Same error happens also if I try to require only the 'epoll' module.

Any idea what's causing this?

Thanks
TG

fivdi, 2015-04-04:
onoff and epoll can only be used to interact with GPIO pins on Linux machines such as the Raspberry Pi and BeagleBone.

What is your project trying to achieve with onoff on a Windows PC?

fivdi, 2015-04-04:
This means that Node.js should be installed on the Raspberry Pi or BeagleBone and then onoff is installed on the Raspberry Pi or BeagleBone.

ThumbGen, 2015-04-04:
Yeah, just tested the project few minutes ago on raspi and it works perfectly...

I am using a WindowsPC with VisualStudio (and nodejs tools for VS) for development and then I push the files to the raspi. I will just mock the gpio access layer during developing on Win and use the onoff library when running on raspi.

Sorry for not realizing sooner my mistake, I guess the 'ticket' can be closed.

PS: Amazing lib BTW, congrats and thank you for providing it!

closed 2015-04-04 by ThumbGen

fivdi, 2015-04-04:
No problem and thanks :)
Good luck with your project.

adityapatadia, 2015-09-07:
I am trying to install epoll in Mac OS X and getting this same error. Does this mean, I cant include this library in Mac OS X? I am using Node.js v0.12

fivdi, 2015-09-07:
This is correct. The epoll module relies on the Linux epoll event notification facility which isn't available on OS X.

FullR, 2015-12-07:
I'm experiencing this error on a Beaglebone Black running Debian 8.1:

```
smartpumps@beaglebone:~$ sudo node test.js
/home/smartpumps/octalbonescript/node_modules/epoll/node_modules/bindings/bindings.js:83
        throw e
              ^
Error: Module did not self-register.
    at Error (native)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at bindings (/home/smartpumps/octalbonescript/node_modules/epoll/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/smartpumps/octalbonescript/node_modules/epoll/epoll.js:1:99)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
```
Any suggestions?

fivdi, 2015-12-07:
This may be happening because epoll (or octalbonescript in this case) was installed and compiled with one version of Node.js but a different version of Node.js is being used at runtime.

It may also be happening because user smartpumps and root have access to different versions of Node.js. (calling `sudo node test.js` means that user root is executing the code and root may have a different PATH to smartpumps.)

To determine whether there is a fundamental problem with epoll perform the following test in an empty directory.
```bash
npm install epoll
cd node_modules/epoll/test/
./runtests
```
If the output of the tests looks similar to the following, everything is ok with epoll, and chances are something is configured incorrectly on the system:
```
started  - do-nothing
finished - do-nothing
started  - do-almost-nothing
finished - do-almost-nothing
started  - verify-events
finished - verify-events
started  - closed
finished - closed
started  - one-shot
finished - one-shot
started  - two-shot
finished - two-shot
started  - performance-check
           11881 events per second
finished - performance-check
started  - no-gc-allowed
finished - no-gc-allowed
```

adityapatadia, 2015-12-12:
Let me know what is result of it. If any update is needed on OBS part, I will do it.