fivdi/epoll #14: can't install epoll on Raspi1 B

brunosantos, 2015-05-17:
I'm trying to install [epoll] on my pi(RPi 1 model B 700Mz 256MB).
My node version is:v0.12.0

I'm running this command:

    sudo npm install --unsafe-perm epoll

and getting back this error:

    > epoll@0.1.13 install /home/pi/node_modules/epoll
    > node-gyp rebuild
    
    child_process: customFds option is deprecated, use stdio instead.
    gyp: /root/.node-gyp/0.12.0/common.gypi not found (cwd: /home/pi/node_modules/epoll) while reading includes of binding.gyp while trying to load binding.gyp
    gyp ERR! configure error
    gyp ERR! stack Error: `gyp` failed with exit code: 1
    gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:343:16)
    gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
    gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
    gyp ERR! System Linux 3.12.26-rpi-aufs
    gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
    gyp ERR! cwd /home/pi/node_modules/epoll
    gyp ERR! node -v v0.12.0
    gyp ERR! node-gyp -v v1.0.2
    gyp ERR! not ok
    npm ERR! Linux 3.12.26-rpi-aufs
    npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "--unsafe-perm" "epoll"
    npm ERR! node v0.12.0
    npm ERR! npm  v2.5.1
    npm ERR! code ELIFECYCLE
    
    npm ERR! epoll@0.1.13 install: `node-gyp rebuild`
    npm ERR! Exit status 1
    npm ERR!
    npm ERR! Failed at the epoll@0.1.13 install script 'node-gyp rebuild'.
    npm ERR! This is most likely a problem with the epoll package,
    npm ERR! not with npm itself.
    npm ERR! Tell the author that this fails on your system:
    npm ERR!     node-gyp rebuild
    npm ERR! You can get their info via:
    npm ERR!     npm owner ls epoll
    npm ERR! There is likely additional logging output above.
    
    npm ERR! Please include the following file with any support request:
    npm ERR!     /home/pi/npm-debug.log

Any idea why?
Note: All this works fine in my RPi2 model B.

fivdi, 2015-05-18:
Part of the error message is the following:

```
gyp: /root/.node-gyp/0.12.0/common.gypi not found (cwd: /home/pi/node_modules/epoll) while reading includes of binding.gyp while trying to load binding.gyp
```

Does file `/root/.node-gyp/0.12.0/common.gypi` exist?
Does directory `/root/.node-gyp/0.12.0` exist?

fivdi, 2015-05-19:
I just successfully installed Node.js v0.12.1 and epoll v0.1.13 on a Raspberry Pi 1 B with 256MB without issues. The Operating System being used is Raspbian 2015-05-05. Here's the output of the install:

```
pi@raspberrypi ~/dev/epoll $ npm install epoll
|
> epoll@0.1.13 install /home/pi/dev/epoll/node_modules/epoll
> node-gyp rebuild

child_process: customFds option is deprecated, use stdio instead.
make: Entering directory '/home/pi/dev/epoll/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  SOLINK_MODULE(target) Release/obj.target/epoll.node: Finished
  COPY Release/epoll.node
make: Leaving directory '/home/pi/dev/epoll/node_modules/epoll/build'
epoll@0.1.13 node_modules/epoll
├── bindings@1.2.1
└── nan@1.8.4
pi@raspberrypi ~/dev/epoll $ node -v
v0.12.1
pi@raspberrypi ~/dev/epoll $ uname -a
Linux raspberrypi 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l GNU/Linux
pi@raspberrypi ~/dev/epoll $ 
```

It's possible to simulate the error by deleting the file `/home/pi/.node-gyp/0.12.1/common.gypi`. This probably means that there's something wrong with the installation. I'd suggest removing the directory `/home/pi/.node-gyp/0.12.1` and all it's content and then try installing epoll again. Note that the first time an addon like epoll is being installed, it can take quite a while for node-gyp to download and decompress what it needs, so patience is required :).

dansitu, 2015-05-20:
I was unable to install epoll on a Pi 2 B with node 0.12.1 and a freshly installed OS with this build:

`Linux raspberrypi 3.18.11-v7+ #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015 armv7l GNU/Linux`

I was connecting over wifi, with an adapter using the mt7061 chipset. I tried numerous times.

I just attempted to install epoll on another fresh install while attached to the network via ethernet and it worked perfectly first time.

It looks like perhaps some of the files get silently truncated during download, maybe because of the crap mt7061 driver. @brunosantos , maybe give this a try connected to wifi?

The only other differences between attempts (it seems unlikely that they are the cause):
- The first time around I had `gpio-admin` installed, the second time I didn't
- The first time around I was installing from within in a `tmux` session, the second time I wasn't

brunosantos, 2015-05-21:
Hello, thanks for your quick help.
thanks @dansitu but I'm hard-wired.
I'm still having issues with it after deleting the node-gyp folder.

first I tried to install node-gyp:

     sudo npm install -g --unsafe-perm node-gyp
    +-- tar@1.0.3 (inherits@2.0.1, block-stream@0.0.7)
    +-- glob@4.5.3 (inherits@2.0.1, inflight@1.0.4, once@1.3.2, minimatch@2.0.8)
    +-- npmlog@1.2.1 (ansi@0.3.0, gauge@1.2.0, are-we-there-yet@1.0.4)
    +-- request@2.55.0 (caseless@0.9.0, aws-sign2@0.5.0, forever-agent@0.6.1, stringstream@0.0.4, oauth-sign@0.6.0, tunnel-agent@0.4.0, isstream@0.1.2, json-stringify-safe@5.0.1, node-uuid@1.4.3, qs@2.4.2, combined-stream@0.0.7, form-data@0.2.0, mime-types@2.0.11, bl@0.9.4, http-signature@0.10.1, tough-cookie@1.1.0, hawk@2.3.1, har-validator@1.7.0)

that looked fine.
Then when I try  npm install epoll, I got:

/
    > epoll@0.1.13 install /home/pi/node_modules/epoll
    > node-gyp rebuild

    gyp WARN install got an error, rolling back install
    gyp ERR! configure error
    gyp ERR! stack Error: socket hang up
    gyp ERR! stack     at createHangUpError (_http_client.js:215:15)
    gyp ERR! stack     at Socket.socketOnEnd (_http_client.js:293:23)
    gyp ERR! stack     at Socket.emit (events.js:129:20)
    gyp ERR! stack     at _stream_readable.js:908:16
    gyp ERR! stack     at process._tickCallback (node.js:355:11)
    gyp ERR! System Linux 3.12.26-rpi-aufs
    gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
    gyp ERR! cwd /home/pi/node_modules/epoll
    gyp ERR! node -v v0.12.0
    gyp ERR! node-gyp -v v1.0.2
    gyp ERR! not ok
    npm ERR! Linux 3.12.26-rpi-aufs
    npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "epoll"
    npm ERR! node v0.12.0
    npm ERR! npm  v2.5.1
    npm ERR! code ELIFECYCLE

    npm ERR! epoll@0.1.13 install: `node-gyp rebuild`
    npm ERR! Exit status 1
    npm ERR!
    npm ERR! Failed at the epoll@0.1.13 install script 'node-gyp rebuild'.
    npm ERR! This is most likely a problem with the epoll package,
    npm ERR! not with npm itself.
    npm ERR! Tell the author that this fails on your system:
    npm ERR!     node-gyp rebuild
    npm ERR! You can get their info via:
    npm ERR!     npm owner ls epoll
    npm ERR! There is likely additional logging output above.

the npm-debug.log looks like:

    249 info install epoll@0.1.13
    250 verbose unsafe-perm in lifecycle true
    251 info epoll@0.1.13 Failed to exec install script
    252 verbose unlock done using /home/pi/.npm/_locks/epoll-aa577be478f2bfb6.lock for /home/pi/node_modules/epoll
    253 verbose stack Error: epoll@0.1.13 install: `node-gyp rebuild`
    253 verbose stack Exit status 1
    253 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/lib/utils/lifecycle.js:213:16)
    253 verbose stack     at EventEmitter.emit (events.js:110:17)
    253 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/lib/utils/spawn.js:14:12)
    253 verbose stack     at ChildProcess.emit (events.js:110:17)
    253 verbose stack     at maybeClose (child_process.js:1008:16)
    253 verbose stack     at Process.ChildProcess._handle.onexit (child_process.js:1080:5)
    254 verbose pkgid epoll@0.1.13
    255 verbose cwd /home/pi
    256 error Linux 3.12.26-rpi-aufs
    257 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "epoll"
    258 error node v0.12.0
    259 error npm  v2.5.1
    260 error code ELIFECYCLE
    261 error epoll@0.1.13 install: `node-gyp rebuild`
    261 error Exit status 1
    262 error Failed at the epoll@0.1.13 install script 'node-gyp rebuild'.
    262 error This is most likely a problem with the epoll package,
    262 error not with npm itself.
    262 error Tell the author that this fails on your system:
    262 error     node-gyp rebuild
    262 error You can get their info via:
    262 error     npm owner ls epoll
    262 error There is likely additional logging output above.
    263 verbose exit [ 1, true ]

Even if this was an issue with my current installation. I've fashed my card with a new Raspian and I get the same problem.
I can install other npm packages but I can't manage to get epoll... which is a dependency of onoff which is the one that I was trying to get originally.

Any thoughts? 
Thanks.

dansitu, 2015-05-21:
Interesting. I'm still thinking this is a connectivity issue - your log
indicates a "socket hang up". Why does the epoll build process connect to
the web beyond what npm does?

One thing that comes to mind is the Pi doesn't have a hardware clock. If
the clock drifts sufficiently there could be problems with any SSL
connection:

http://security.stackexchange.com/questions/72866/what-role-does-clock-synchronization-play-in-ssl-communcation

Seems unlikely that this would cause the file truncation I experienced, and
I'd expect a different error message than the one you experienced, but
perhaps there's another time sensitive protocol involved. You could try
synching with an NTP server immediately before running npm install?

I was trying to install onoff, too. It eventually worked.

brunosantos, 2015-05-21:
This might seem as a silly question but. Can I install epoll as global (-g)?
because I get a different error....

	sudo npm install -g epoll
	|
	> epoll@0.1.13 install /usr/local/lib/node_modules/epoll
	> node-gyp rebuild

	gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/0.12.0"
	gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/node_modules/epoll/.node-gyp"
	gyp WARN install got an error, rolling back install
	gyp WARN install got an error, rolling back install
	gyp ERR! configure error
	gyp ERR! stack Error: node-v0.12.0.tar.gz local checksum 8ecb2cd010d583c72f5993a67fc6dc2ae6321532aca7936939b17a73f4a873db not match remote 9700e23af4e9b3643af48cef5f2ad20a1331ff531a12154eef2bfb0bb1682e32
	gyp ERR! stack     at deref (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:299:20)
	gyp ERR! stack     at IncomingMessage.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:340:13)
	gyp ERR! stack     at IncomingMessage.emit (events.js:129:20)
	gyp ERR! stack     at _stream_readable.js:908:16
	gyp ERR! stack     at process._tickCallback (node.js:355:11)
	gyp ERR! System Linux 3.12.26-rpi-aufs
	gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
	gyp ERR! cwd /usr/local/lib/node_modules/epoll
	gyp ERR! node -v v0.12.0
	gyp ERR! node-gyp -v v1.0.2
	gyp ERR! not ok
	npm ERR! Linux 3.12.26-rpi-aufs
	npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "-g" "epoll"
	npm ERR! node v0.12.0
	npm ERR! npm  v2.5.1
	npm ERR! code ELIFECYCLE

	npm ERR! epoll@0.1.13 install: `node-gyp rebuild`
	npm ERR! Exit status 1
	npm ERR!
	npm ERR! Failed at the epoll@0.1.13 install script 'node-gyp rebuild'.
	npm ERR! This is most likely a problem with the epoll package,
	npm ERR! not with npm itself.
	npm ERR! Tell the author that this fails on your system:
	npm ERR!     node-gyp rebuild
	npm ERR! You can get their info via:
	npm ERR!     npm owner ls epoll
	npm ERR! There is likely additional logging output above.

Also...I'm executing the commands via ssh using screen....could that be an issue?
Thanks for the replies

dansitu, 2015-05-21:
I had the same error trying to install globally; not sure of an explanation.

Could try running `npm install` using the bare terminal rather than
`screen`? My original issues occurred within a `tmux` session (similar to
`screen`) so maybe there's some weird issue there. I did everything over
SSH, including when it worked.

fivdi, 2015-05-21:
If a native module like epoll is installed with root privileges (sudo), gyp recompiles as "nobody", not as "root". This can be avoided with option --unsafe-pern

Note that the checksum error this time is yet another indication that there is something wrong with the internet connection. Each time there is a different error.

Is it possible to install any native modules? For example, does `npm install serialport` work?

fivdi, 2015-05-21:
Is the Raspberry Pi being overclocked?
Has the same SD card worked successfully on the same Raspberry Pi in the past?

fivdi, 2015-05-22:
When a native module is installed, gyp downloads and decompresses the node tar file. This is why there is a lot more internet access when the first native module is installed.

Does `dmesg` display any error messages directly after an unsuccessful install?

brunosantos, 2015-05-25:
pi is not overclocked (but good call).
When I run:
 	cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq
	700000

I tried to install serialport with no luck:

		|
	> serialport@1.7.1 install /home/pi/node_modules/serialport
	> node-pre-gyp install --fallback-to-build

	gyp WARN install got an error, rolling back install
	gyp ERR! configure error
	gyp ERR! stack Error: node-v0.12.0.tar.gz local checksum f47aacb7817286b5cfc457fd0f0b0eec54b5898fb774cc9d8634acea86db767d not match remote 9700e23af4e9b3643af48cef5f2ad20a1331ff531a12154eef2bfb0bb1682e32
	gyp ERR! stack     at deref (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:299:20)
	gyp ERR! stack     at IncomingMessage.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:340:13)
	gyp ERR! stack     at IncomingMessage.emit (events.js:129:20)
	gyp ERR! stack     at _stream_readable.js:908:16
	gyp ERR! stack     at process._tickCallback (node.js:355:11)
	gyp ERR! System Linux 3.12.26-rpi-aufs
	gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/home/pi/node_modules/serialport/build/serialport/v1.7.1/Release/node-v14-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/home/pi/node_modules/serialport/build/serialport/v1.7.1/Release/node-v14-linux-arm"
	gyp ERR! cwd /home/pi/node_modules/serialport
	gyp ERR! node -v v0.12.0
	gyp ERR! node-gyp -v v1.0.2
	gyp ERR! not ok
	node-pre-gyp ERR! build error
	node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/pi/node_modules/serialport/build/serialport/v1.7.1/Release/node-v14-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/serialport/v1.7.1/Release/node-v14-linux-arm' (1)
	node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
	node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
	node-pre-gyp ERR! stack     at maybeClose (child_process.js:1008:16)
	node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1080:5)
	node-pre-gyp ERR! System Linux 3.12.26-rpi-aufs
	node-pre-gyp ERR! command "node" "/home/pi/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
	node-pre-gyp ERR! cwd /home/pi/node_modules/serialport
	node-pre-gyp ERR! node -v v0.12.0
	node-pre-gyp ERR! node-pre-gyp -v v0.6.7
	node-pre-gyp ERR! not ok
	Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/pi/node_modules/serialport/build/serialport/v1.7.1/Release/node-v14-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/serialport/v1.7.1/Release/node-v14-linux-arm' (1)
	npm ERR! Linux 3.12.26-rpi-aufs
	npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport"
	npm ERR! node v0.12.0
	npm ERR! npm  v2.5.1
	npm ERR! code ELIFECYCLE

	npm ERR! serialport@1.7.1 install: `node-pre-gyp install --fallback-to-build`
	npm ERR! Exit status 1
	npm ERR!
	npm ERR! Failed at the serialport@1.7.1 install script 'node-pre-gyp install --fallback-to-build'.
	npm ERR! This is most likely a problem with the serialport package,
	npm ERR! not with npm itself.
	npm ERR! Tell the author that this fails on your system:
	npm ERR!     node-pre-gyp install --fallback-to-build
	npm ERR! You can get their info via:
	npm ERR!     npm owner ls serialport
	npm ERR! There is likely additional logging output above.

	npm ERR! Please include the following file with any support request:
	npm ERR!     /home/pi/npm-debug.log

fivdi, 2015-05-26:
Has the same SD card worked successfully on the same Raspberry Pi in the past?
What is the output of `dmesg` after an unsuccessful install?

brunosantos, 2015-05-26:
I should have mentioned that I'm using [berryboot][2] for multiple distribs.
I also have berryboot with my raspi 2 and that works fine.
I was never able to install onoff, epoll or serialport in my raspi1 with this SD card.
Since it's an SD card I can't test it in my raspi2. But I think (because I'm away atm) the SD card is:
SDSDX-016G-X46

I was able to install [rpi-gpio][1] and to turn a led on and off with node (not sure if that is a native module).

Here is the dmesg:

	$ dmesg
	[    0.000000] Booting Linux on physical CPU 0x0
	[    0.000000] Initializing cgroup subsys cpu
	[    0.000000] Initializing cgroup subsys cpuacct
	[    0.000000] Linux version 3.12.26-rpi-aufs (max@lynx) (gcc version 4.5.3 (Buildroot 2012.05-g41d6ffe-dirty) ) #1 PREEMPT Thu Aug 14 10:48:01 CEST 2014
	[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
	[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
	[    0.000000] Machine: BCM2708
	[    0.000000] early_vc_cma_mem(0/0x5c00000@0x9000000)
	[    0.000000]  -> initial 0, size 5c00000, base 9000000<6>[    0.000000] cma: CMA: reserved 92 MiB at 09000000
	[    0.000000] cma: CMA: reserved 8 MiB at 06c00000
	[    0.000000] Memory policy: ECC disabled, Data cache writeback
	[    0.000000] On node 0 totalpages: 56320
	[    0.000000] free_area_init_node: node 0, pgdat c06665a0, node_mem_map c0717000
	[    0.000000]   Normal zone: 472 pages used for memmap
	[    0.000000]   Normal zone: 0 pages reserved
	[    0.000000]   Normal zone: 56320 pages, LIFO batch:15
	[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
	[    0.000000] pcpu-alloc: [0] 0
	[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 55848
	[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708.boardrev=0x2 bcm2708.serial=0xf648380c smsc95xx.macaddr=B8:27:EB:48:38:0C sdhci-bcm2708.emmc_clock_freq=250000000 vc-cma-mem=0/0x5c00000@0x9000000 mem=0x8000000@0x0 mem=0x5c00000@0x9000000 vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000  smsc95xx.turbo_mode=N elevator=deadline quiet bootmenutimeout=10 datadev=mmcblk0p2 qmap=gb ipv4=192.168.1.42/255.255.255.0/192.168.1.1
	[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
	[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
	[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
	[    0.000000] Memory: 105068K/225280K available (4714K kernel code, 251K rwdata, 1404K rodata, 148K init, 703K bss, 120212K reserved)
	[    0.000000] Virtual kernel memory layout:
	[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
	[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
	[    0.000000]     vmalloc : 0xcf000000 - 0xff000000   ( 768 MB)
	[    0.000000]     lowmem  : 0xc0000000 - 0xcec00000   ( 236 MB)
	[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
	[    0.000000]       .text : 0xc0008000 - 0xc0601d7c   (6120 kB)
	[    0.000000]       .init : 0xc0602000 - 0xc0627134   ( 149 kB)
	[    0.000000]       .data : 0xc0628000 - 0xc0666ea0   ( 252 kB)
	[    0.000000]        .bss : 0xc0666eac - 0xc0716dc8   ( 704 kB)
	[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
	[    0.000000] Preemptible hierarchical RCU implementation.
	[    0.000000] NR_IRQS:394
	[    0.000000] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 4294967ms
	[    0.000000] Switching to timer-based delay loop
	[    0.000000] Console: colour dummy device 80x30
	[    0.000000] console [tty0] enabled
	[    0.000550] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
	[    0.000571] pid_max: default: 32768 minimum: 301
	[    0.000980] Mount-cache hash table entries: 512
	[    0.001790] Initializing cgroup subsys memory
	[    0.001865] Initializing cgroup subsys devices
	[    0.001883] Initializing cgroup subsys freezer
	[    0.001896] Initializing cgroup subsys blkio
	[    0.002014] CPU: Testing write buffer coherency: ok
	[    0.002488] Setting up static identity map for 0xc047b508 - 0xc047b564
	[    0.004246] devtmpfs: initialized
	[    0.021528] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
	[    0.033942] NET: Registered protocol family 16
	[    0.039302] DMA: preallocated 4096 KiB pool for atomic coherent allocations
	[    0.039937] cpuidle: using governor ladder
	[    0.039961] cpuidle: using governor menu
	[    0.040645] bcm2708.uart_clock = 0
	[    0.042571] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
	[    0.042590] hw-breakpoint: maximum watchpoint size is 4 bytes.
[... 144 lines cut ...]
	[    3.971078] EXT4-fs (mmcblk0p2): INFO: recovery required on readonly filesystem
	[    3.971112] EXT4-fs (mmcblk0p2): write access will be enabled during recovery
	[    4.623127] EXT4-fs (mmcblk0p2): orphan cleanup on readonly fs
	[    4.624923] EXT4-fs (mmcblk0p2): 3 orphan inodes deleted
	[    4.624949] EXT4-fs (mmcblk0p2): recovery complete
	[    4.854092] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
	[    4.882024] EXT4-fs (mmcblk0p2): re-mounted. Opts: data=ordered
	[    5.485539] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x45E1
	[   15.118072] bcm2835-cpufreq: switching to governor ondemand
	[   15.118104] bcm2835-cpufreq: switching to governor ondemand
[... cut ...]
	[   19.173715] usbcore: registered new interface driver uvcvideo
	[   19.173737] USB Video Class driver (1.1.1)
	[   20.004889] bcm2708-i2s bcm2708-i2s.0: Failed to create debugfs directory
	[   25.122805] 4:3:1: cannot get freq at ep 0x82
	[   25.130450] usbcore: registered new interface driver snd-usb-audio
	[   43.685841] bcm2835-cpufreq: switching to governor ondemand
	[   43.685874] bcm2835-cpufreq: switching to governor ondemand

 [1]: https://www.npmjs.com/package/rpi-gpio
[2]: http://www.berryterminal.com/doku.php/berryboot

fivdi, 2015-05-26:
Is the `dmesg` output posted above the output after an unsuccessful install of `epoll` or `serialport`?

`rpi-gpio` isn't a native module, so there's much less network traffic when it's installed. `epoll` and `serialport` are native modules which means that there's a lot more network traffic when the first native module is installed. `node-v0.12.0.tar.gz` is downloaded and decompressed. What can be seen from the error messages that you have already gotten is that `node-v0.12.0.tar.gz` can't be downloaded successfully.

Once there was this error:
```
gyp ERR! stack Error: node-v0.12.0.tar.gz local checksum 8ecb2cd010d583c72f5993a67fc6dc2ae6321532aca7936939b17a73f4a873db not match remote 9700e23af4e9b3643af48cef5f2ad20a1331ff531a12154eef2bfb0bb1682e32
```
and once there was this error:
```
gyp ERR! stack Error: node-v0.12.0.tar.gz local checksum f47aacb7817286b5cfc457fd0f0b0eec54b5898fb774cc9d8634acea86db767d not match remote 9700e23af4e9b3643af48cef5f2ad20a1331ff531a12154eef2bfb0bb1682e32
```
As can be seen from the error messages, there are issues with the checksum for `node-v0.12.0.tar.gz`. You need to figure out what the issue is here. It has nothing to do with `epoll` or `serialport`. I have never used berryboat, so I don't know if it's responsible for or plays a role in the issue. 

I'm afraid I'm out of ideas for helping out now so I'm going to close the issue as it's not a problem with `epoll`. Feel free to create a new issue if a concrete problem with epoll is found.

closed 2015-05-26 by fivdi

fivdi, 2015-05-26:
@brunosantos I just took a closer look at the `dmesg` output. The following messages indicate that there is something wrong with the SD card:

```
[    3.971078] EXT4-fs (mmcblk0p2): INFO: recovery required on readonly filesystem
[    3.971112] EXT4-fs (mmcblk0p2): write access will be enabled during recovery
[    4.623127] EXT4-fs (mmcblk0p2): orphan cleanup on readonly fs
[    4.624923] EXT4-fs (mmcblk0p2): 3 orphan inodes deleted
[    4.624949] EXT4-fs (mmcblk0p2): recovery complete
[    4.854092] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    4.882024] EXT4-fs (mmcblk0p2): re-mounted. Opts: data=ordered
```

brunosantos, 2015-05-28:
Thanks for the help @fivdi . yes I agree it's not a problem with epoll itself.

I ran the dmesg again after npm install serialport and noticed this:
	[98759.503781] Out of memory: Kill process 14218 (npm) score 246 or sacrifice child

could it be lack of memory during the install process?

here is the full dmesg:

	[    0.000000] Booting Linux on physical CPU 0x0
	[    0.000000] Initializing cgroup subsys cpu
	[    0.000000] Initializing cgroup subsys cpuacct
	[    0.000000] Linux version 3.12.26-rpi-aufs (max@lynx) (gcc version 4.5.3 (Buildroot 2012.05-g41d6ffe-dirty) ) #$
	[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
	[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
	[    0.000000] Machine: BCM2708
	[    0.000000] early_vc_cma_mem(0/0x5c00000@0x9000000)
	[    0.000000]  -> initial 0, size 5c00000, base 9000000<6>[    0.000000] cma: CMA: reserved 92 MiB at 09000000
	[    0.000000] cma: CMA: reserved 8 MiB at 06c00000
	[    0.000000] Memory policy: ECC disabled, Data cache writeback
	[    0.000000] On node 0 totalpages: 56320
	[    0.000000] free_area_init_node: node 0, pgdat c06665a0, node_mem_map c0717000
	[    0.000000]   Normal zone: 472 pages used for memmap
	[    0.000000]   Normal zone: 0 pages reserved
	[    0.000000]   Normal zone: 56320 pages, LIFO batch:15
	[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
	[    0.000000] pcpu-alloc: [0] 0
	[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 55848
	[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708.boa$
	[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
	[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
	[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
	[    0.000000] Memory: 105068K/225280K available (4714K kernel code, 251K rwdata, 1404K rodata, 148K init, 703K bs$
	[    0.000000] Virtual kernel memory layout:
	[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
	[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
	[    0.000000]     vmalloc : 0xcf000000 - 0xff000000   ( 768 MB)
	[    0.000000]     lowmem  : 0xc0000000 - 0xcec00000   ( 236 MB)
	[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
	[    0.000000]       .text : 0xc0008000 - 0xc0601d7c   (6120 kB)
	[    0.000000]       .init : 0xc0602000 - 0xc0627134   ( 149 kB)
	[    0.000000]       .data : 0xc0628000 - 0xc0666ea0   ( 252 kB)
	[    0.000000]        .bss : 0xc0666eac - 0xc0716dc8   ( 704 kB)
	[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
	[    0.000000] Preemptible hierarchical RCU implementation.
	[    0.000000] NR_IRQS:394
	[    0.000000] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 4294967ms
	[    0.000000] Switching to timer-based delay loop
	[    0.000000] Console: colour dummy device 80x30
	[    0.000000] console [tty0] enabled
	[    0.000550] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
	[    0.000571] pid_max: default: 32768 minimum: 301
	[    0.000980] Mount-cache hash table entries: 512
	[    0.001790] Initializing cgroup subsys memory
	[    0.001865] Initializing cgroup subsys devices
	[    0.001883] Initializing cgroup subsys freezer
	[    0.001896] Initializing cgroup subsys blkio
	[    0.002014] CPU: Testing write buffer coherency: ok
	[    0.002488] Setting up static identity map for 0xc047b508 - 0xc047b564
	[    0.004246] devtmpfs: initialized
	[    0.021528] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
	[    0.033942] NET: Registered protocol family 16
	[    0.039302] DMA: preallocated 4096 KiB pool for atomic coherent allocations
	[    0.039937] cpuidle: using governor ladder
	[    0.039961] cpuidle: using governor menu
	[    0.040645] bcm2708.uart_clock = 0
	[    0.042571] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
	[    0.042590] hw-breakpoint: maximum watchpoint size is 4 bytes.
	[    0.042603] mailbox: Broadcom VideoCore Mailbox driver

[... 340 lines cut ...]
	[98759.558485] [14202]  1000 14202     1412      316       5        0             0 bash
	[98759.558504] [14218]  1000 14218    27156    13230      66        0             0 npm
	[98759.558526] [14253]  1000 14253    19758     5099      41        0             0 node-pre-gyp
	[98759.558545] [14328]  1000 14328    20454     4375      41        0             0 node-gyp
	[98759.558565] [14403]     0 14403     1504       94       5        0             0 sshd
	[98759.558583] [14404]     0 14404     1504       92       6        0             0 sshd
	[98759.558600] [14405]     0 14405     1483       57       5        0             0 sshd
	[98759.558636] [14406]     0 14406     1483       57       5        0             0 sshd
	[98759.558658] [14407]     0 14407     1059       47       5        0             0 cron
	[98759.558677] [14408]     0 14408     1483       53       5        0             0 sshd
	[98759.558695] [14409]     0 14409     1483       53       5        0             0 sshd
	[98759.558713] [14410]     0 14410     1483       53       5        0             0 sshd
	[98759.558730] [14411]     0 14411     1483       53       5        0             0 sshd
	[98759.558748] [14412]     0 14412     1483       53       5        0             0 sshd
[... cut ...]
	[98759.558872] [14418]     0 14418     1483       52       5        0             0 sshd
	[98759.558890] [14419]     0 14419     1483       52       5        0             0 sshd
	[98759.558910] [14420]     0 14420     1483       52       5        0             0 sshd
	[98759.558929] [14421]     0 14421     1482       50       4        0             0 sshd
	[98759.558942] Out of memory: Kill process 14218 (npm) score 246 or sacrifice child
	[98759.558962] Killed process 14218 (npm) total-vm:108624kB, anon-rss:52912kB, file-rss:8kB

fivdi, 2015-05-28:
I guess it could be a lack of memory and that appears to be the case here. Why it happens is the question. Normally, I wouldn't expect `npm install serialport` to result in an `Out of memory` error, even if there isn't enough RAM. I'd expect Raspbian to use the swap file to avoid the `Out of memory` error. Does your Pi have a swap file? Check with the command `swapon -s`.

At boot, the dmesg output does say `Memory: 105068K/225280K available`. 105068K of 225280K available seems low to me. What's all the memory being used for? I see `Memory: 226148K/245760K` on boot on my Pi 1 256MB.