fivdi/epoll #16: Does not compile on RasPi2

WilliWichtel, 2015-09-21:
Hi,

unfortunately epoll does not compile on my Pi2 running raspbian. The solution from #14 does not work for me. >Could someone give me a hint?
Here is the shortened output:

falk@raspberrypi ~/arvid $ sudo npm install --unsafe-perm epoll
/
> epoll@0.1.15 install /home/falk/arvid/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/falk/arvid/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../src/epoll.cc:11:0:
/root/.node-gyp/4.1.0/include/node/v8.h:336:1: error: expected unqualified-id before ‘using’
/root/.node-gyp/4.1.0/include/node/v8.h:469:1: error: expected unqualified-id before ‘using’
/root/.node-gyp/4.1.0/include/node/v8.h:852:1: error: expected unqualified-id before ‘using’
In file included from ../node_modules/nan/nan.h:182:0,
                 from ../src/epoll.cc:15:
../node_modules/nan/nan_maybe_43_inl.h:13:1: error: expected unqualified-id before ‘using’
../node_modules/nan/nan_maybe_43_inl.h:16:1: error: expected unqualified-id before ‘using’
../node_modules/nan/nan_maybe_43_inl.h:19:12: error: ‘Maybe’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:24:12: error: ‘Maybe’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:31:1: error: ‘MaybeLocal’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:36:1: error: ‘MaybeLocal’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:41:1: error: ‘Maybe’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:46:1: error: ‘MaybeLocal’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:51:1: error: ‘MaybeLocal’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:60:1: error: ‘MaybeLocal’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:65:12: error: ‘MaybeLocal’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:70:12: error: ‘Maybe’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:77:12: error: ‘Maybe’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:84:12: error: ‘Maybe’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:92:12: error: ‘MaybeLocal’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:99:1: error: ‘MaybeLocal’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:109:12: error: ‘Maybe’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:115:12: error: ‘Maybe’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:119:12: error: ‘Maybe’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:126:1: error: ‘Maybe’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:131:1: error: ‘MaybeLocal’ does not name a type
../node_modules/nan/nan_maybe_43_inl.h:136:1: error: ‘MaybeLocal’ does not name a type
...
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-unused-local-typedefs" [enabled by default]
epoll.target.mk:90: recipe for target 'Release/obj.target/epoll/src/epoll.o' failed
make: *** [Release/obj.target/epoll/src/epoll.o] Error 1
make: Leaving directory '/home/falk/arvid/node_modules/epoll/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.1.6-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/falk/arvid/node_modules/epoll
gyp ERR! node -v v4.1.0
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok 
npm ERR! Linux 4.1.6-v7+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "--unsafe-perm" "epoll"
npm ERR! node v4.1.0
npm ERR! npm  v2.14.3
npm ERR! code ELIFECYCLE

npm ERR! epoll@0.1.15 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.15 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/falk/arvid/npm-debug.log
falk@raspberrypi ~/arvid $

fivdi, 2015-09-21:
To install native addons like epoll with Node.js v4.1 gcc/g++ 4.8 is needed. It looks like gcc/g++ 4.6 is installed on your system. Check the gcc version with the following command:
```
gcc --version
```
If gcc 4.8 is not installed, please install gcc/g++ 4.8 as described [here](https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons) and then try to install epoll again.

WilliWichtel, 2015-09-21:
much better now ;-)
thank you

closed 2015-09-21 by WilliWichtel