fivdi/epoll #17: Epoll module does not install on latest debian image for BeagleBone

adityapatadia, 2015-12-12:
Hi I downloaded latest Debian 8.2 image for BeagleBone Black and installed it. After that, I tried ```npm install -g epoll```. It gives following error.

```

> epoll@0.1.16 install /usr/local/lib/node_modules/epoll
> node-gyp rebuild

gyp ERR! configure error 
gyp ERR! stack Error: EACCES, mkdir '/usr/local/lib/node_modules/epoll/build'
gyp ERR! System Linux 4.1.12-ti-r29
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /usr/local/lib/node_modules/epoll
gyp ERR! node -v v0.10.38
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! epoll@0.1.16 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.16 install script.
npm ERR! This is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.1.12-ti-r29
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "epoll"
npm ERR! cwd /root
npm ERR! node -v v0.10.38
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /root/npm-debug.log
npm ERR! not ok code 0

```

fivdi, 2015-12-12:
This issues is related to how `npm install` handles privileges. Attempting to install any native module globally as root with `-g` will result in the same errors. For example, `npm install -g buffertools` will result in the same errors. To avoid the issue use `npm install --unsafe-perm -g epoll` instead.
```
root@beaglebone:~# uname -a
Linux beaglebone 4.1.12-ti-r29 #1 SMP PREEMPT Mon Nov 9 22:46:19 UTC 2015 armv7l GNU/Linux
root@beaglebone:~# node -v
v0.10.38
root@beaglebone:~# npm -v
1.4.21
root@beaglebone:~# npm install --unsafe-perm -g epoll
-
> epoll@0.1.16 install /usr/local/lib/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/usr/local/lib/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  SOLINK_MODULE(target) Release/obj.target/epoll.node: Finished
  COPY Release/epoll.node
make: Leaving directory '/usr/local/lib/node_modules/epoll/build'
epoll@0.1.16 /usr/local/lib/node_modules/epoll
├── bindings@1.2.1
└── nan@2.1.0
```

adityapatadia, 2015-12-12:
The issue affects all users of OctalBoneScript. Since they install OBS as ``` npm install -g octalbonescript ``` Do you suggest installing OBS with ``` npm install -g --unsafe-perm octalbonescript``` ?

fivdi, 2015-12-12:
If it's necessary to install OBS globally as root I don't think they have another choice. Why is it necessary to install globally? Would it not be better to recommend `npm install octalbonescript` instead?

adityapatadia, 2015-12-12:
I think you are correct. I should recommend them to install on per project basis with package.json files. I will also write that if someone wants to install globally, how to do it. 

Thanks for help.

closed 2015-12-12 by adityapatadia

fivdi, 2015-12-12:
Your welcome :)