fivdi/epoll #19: Cannot install rpi-gpio using Docker using: HypriotOS (Host), Raspbian (Virtual OS), node v4.2.1, npm v3.9.5

steam0, 2016-06-08:
npm ERR! Linux 4.4.10-hypriotos-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
npm ERR! node v4.2.1
npm ERR! npm  v3.9.5
npm ERR! code ELIFECYCLE

npm ERR! epoll@0.1.19 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the epoll@0.1.19 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs epoll
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /src/npm-debug.log

steam0, 2016-06-08:
Adding contents of docker file as well:

`#Download OS
FROM	resin/rpi-raspbian

# Install CURL
RUN	    apt-get update
RUN     apt-get upgrade
RUN 	apt-get install --yes wget

# Install Node
RUN     wget http://node-arm.herokuapp.com/node_latest_armhf.deb
RUN     sudo dpkg -i node_latest_armhf.deb
RUN     node -v

# Install npm
RUN     apt-get install --yes npm
RUN     sudo npm install -g npm@latest
RUN     sudo apt-get install --yes build-essential

# Install app dependencies
COPY 	package.json /src/package.json
RUN 	cd /src; npm install`

The last command in this file generates the error.

fivdi, 2016-06-08:
I'm not familiar with Docker or HypriotOS so probably will not be able to help much.

However, what can be seen is an ENETUNREACH error suggesting that a network is not reachable. epoll doesn't access any networks. What network is not reachable?

Also, is it possible to install other native modules? For example, can buffertools be installed?

steam0, 2016-06-09:
Yes, buffertools can be installed.

fivdi, 2016-06-09:
And the ENETUNREACH error. Any idea where that's coming from?

steam0, 2016-06-09:
I just edited my comment about it 👍

fivdi, 2016-06-09:
Does `npm install epoll` work if called directly from the command line?

epoll doesn't actually know about the GPIO pins on the raspberry pi. It only knows about file descriptors. rpi-gpio knows about them. I don't know if HipriotOS  needs to know about GPIO pins.

steam0, 2016-06-09:
I updated my Dockerfile to contain

`RUN npm install epoll`

before I copy the package.json file and execute npm install. 

That actually fixed all errors. I don't really know how, but it did. Thank you for your time.

fivdi, 2016-06-09:
Is epoll now being installed in a different directory and not below `/src`? If so, I wonder if the initial problem has something to do with permissions.

steam0, 2016-06-09:
This is the updated dockerfile:

```
#Download OS
FROM	resin/rpi-raspbian

# Install CURL
RUN	apt-get update
RUN        apt-get upgrade
RUN 	apt-get install --yes wget

# Install Node
RUN     wget http://node-arm.herokuapp.com/node_latest_armhf.deb
RUN     sudo dpkg -i node_latest_armhf.deb
RUN     node -v

# Install npm
RUN     apt-get install --yes npm
RUN     sudo npm install -g npm@latest
RUN     sudo apt-get install --yes build-essential

# Install epoll seperately. Fixes a bug where npm install fails on rpi-gpio
RUN     npm install epoll

# Install app dependencies
COPY 	package.json /src/package.json
RUN 	cd /src; npm install

# Bundle app source
COPY    app/ /src/
COPY 	server.js /src/

# Expose
EXPOSE 3000
CMD 	["nodejs", "/src/server.js]
```

As we can see, the `npm install epoll` command is not run from inside /src. So yeah, permission errors might be what caused this problem.

fivdi, 2016-06-10:
@steam0 I'm going to close this issue as it was configuration issues and I'm unaware of any modifications that need to be made to epoll. If it turns out that epoll does need to be modified, the issue can be reopened.

closed 2016-06-10 by fivdi