fivdi/epoll #20: epoll.node bindings file unable to locate

hulk-mk, 2016-06-22:
following error coming. I think its issue with the epoll module. I'm using it to write on lcd using lcd module on BBB. Any help will be appreciated.

debugger listening on port 15454

/usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/node_modules/bindings/bindings.js:91
  throw err
        ^
Error: Could not locate the bindings file. Tried:
 → /usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/build/epoll.node
 → /usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/build/Debug/epoll.node
 → /usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/build/Release/epoll.node
 → /usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/out/Debug/epoll.node
 → /usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/Debug/epoll.node
 → /usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/out/Release/epoll.node
 → /usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/Release/epoll.node
 → /usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/build/default/epoll.node
 → /usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/compiled/0.10.29/linux/arm/epoll.node
    at bindings (/usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/usr/local/lib/node_modules/lcd/node_modules/onoff/node_modules/epoll/epoll.js:1:99)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/usr/local/lib/node_modules/lcd/node_modules/onoff/onoff.js:4:11)
    at Module._compile (module.js:456:26)

fivdi, 2016-06-22:
The binary for the epoll package which is called `epoll.node` doesn't appear to exist on the system. The above output doesn't provide any information as to why it doesn't exist.

Can you create a new empty directory in the the home directory, run the command `npm install lcd` in that directory, and post the complete output of that command please?

This is the output I see:
```
root@beaglebone:~/lcd# npm install lcd

> epoll@0.1.19 install /root/lcd/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/root/lcd/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/root/lcd/node_modules/epoll/build'
/root/lcd
└─┬ lcd@1.1.4 
  ├─┬ onoff@1.1.1 
  │ └─┬ epoll@0.1.19 
  │   ├── bindings@1.2.1 
  │   └── nan@2.3.5 
  └── q@1.4.1 

npm WARN enoent ENOENT, open '/root/lcd/package.json'
npm WARN lcd No description
npm WARN lcd No repository field.
npm WARN lcd No README data
npm WARN lcd No license field.
root@beaglebone:~/lcd# 
```

fivdi, 2016-06-25:
@hulk-mk was it possible to successfully install the lcd package?

closed 2016-06-29 by fivdi

bmwertman, 2018-05-18:
I have the same issue and tried your suggestion but I get the below error message after running `npm install lcd` in the new directory created inside `home` directory;

```
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/testEpoll/node_modules/epoll/.node-gyp"
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/testEpoll/node_modules/epoll/.node-gyp/8.11.2"
```

fivdi, 2018-05-18:
@bmwertman  please open a new issue to describe your problem as it looks you are facing a very different issue.

fivdi, 2018-05-18:
Ok, fair enough :smile: we can take a look at the issue here.

Please post the exact command used for the install and the complete output of that command.

bmwertman, 2018-05-18:
Cool thanks :+1: 

Are you talking about the command used to install `epoll`? Becuase I never directly tried to install epoll. 

I'm presuming it's a dependency for your other lib `onoff` which I had been using successfully in my program before my sd card died and I had to reinstall my OS on a new sd card and re-setup my node environment.

fivdi, 2018-05-18:
> Are you talking about the command used to install epoll? Becuase I never directly tried to install epoll.

The command that was used to install what ever that happens to be... The complete command

bmwertman, 2018-05-18:
```
sudo npm install onoff
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

+ onoff@3.1.0
updated 1 package in 6.782s
[+] no known vulnerabilities found [57 packages audited]
```

fivdi, 2018-05-18:
Why is `sudo` being used?

bmwertman, 2018-05-18:
Without sudo I get permission denied;
```
sudo npm install onoff
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

+ onoff@3.1.0
updated 1 package in 6.782s
[+] no known vulnerabilities found [57 packages audited]
```

fivdi, 2018-05-18:
Are you on a Raspberry Pi logged on as user pi in the home directory?
Please post the output of the following command
```
ls -la
```

bmwertman, 2018-05-18:
```
 ls -la
total 28
drwxr-xr-x 4 501 dialout 4096 May 11 16:25 .
drwxr-xr-x 3 pi  pi      4096 May 18 00:18 ..
drwxr-xr-x 2 501 dialout 4096 May  9 00:05 drainTimes
drwxr-xr-x 2 501 dialout 4096 May  9 00:23 fillTimes
-rwxr-xr-x 1 501 dialout 8761 May 15 16:04 reefbot.js
```

bmwertman, 2018-05-18:
I'm on Raspberry Pi. not in the home directory though.

fivdi, 2018-05-18:
In the post above the directory `/home/pi/node_modules` was being as the destination directory for the install. What's the output of the commands
```
cd
ls -l
```

bmwertman, 2018-05-18:
```
pi@raspberrypi:~/Documents/Reef $ cd
pi@raspberrypi:~ $ ls -l
total 16660
drwxr-xr-x  2 pi   pi       4096 May 17 19:24 Desktop
drwxr-xr-x  3 pi   pi       4096 May 18 00:18 Documents
drwxr-xr-x  2 pi   pi       4096 May 18 13:52 Downloads
drwxr-xr-x  2 pi   pi       4096 May 17 19:24 Music
drwxr-xr-x 25 root root     4096 May 18 17:38 node_modules
drwxr-xr-x  6 pi   pi       4096 May 15 18:54 node-v8.11.2-linux-armv7l
-rw-r--r--  1 pi   pi   17004541 May 15 18:55 node-v8.11.2-linux-armv7l.tar.gz
-rw-r--r--  1 root root     5971 May 18 14:38 package-lock.json
drwxr-xr-x  2 pi   pi       4096 May 17 19:24 Pictures
drwxr-xr-x  2 pi   pi       4096 May 17 19:24 Public
drwxr-xr-x  2 pi   pi       4096 Apr 18 00:57 python_games
drwxr-xr-x  2 pi   pi       4096 May 17 19:24 Templates
drwxr-xr-x  2 pi   pi       4096 May 17 19:24 Videos
pi@raspberrypi:~ $ 

```

fivdi, 2018-05-18:
I think your system is a little messed up.
The `node_modules` directory in the home directory is owned by root so user pi can use it.
Why is `node_modules` owned by root?
Is `node_modules` in the home directory really needed?
If it's not needed please remove it and it's content.

bmwertman, 2018-05-18:
Better to delete `node_modules` owned by`root` in `home` directory? or just `sudo chown -R` it? If I delete then I would need to reinstall everything right?

fivdi, 2018-05-18:
Yes, if you delete it you will need to reinstall everything. It there a lot in `node_modules`? You could also change the owner. That being said, the home directory is sort of a strange place of a `node_modules` directory in my opinion.

bmwertman, 2018-05-18:
I'm a little new to working on the RPi. On my Mac I typically just set node and my node_modules up globally using the `-g` flag with npm. Can you recommend a better structure for the RPi

fivdi, 2018-05-18:
My rules are:
- Never install globally with -g
- Never install with sudo
- Never install in the home directory
- Create a new directory for each project (as user pi)
- cd into that new directory
- Install whatever is needed using `npm install ...` (as user pi without sudo)

bmwertman, 2018-05-18:
Cool thanks @fivdi! I'll try that.

fivdi, 2018-05-18:
Are you going to remove `/home/pi/node_modules`?

bmwertman, 2018-05-18:
Yeah I haven't got that deep into setting things back up. I'll tear it all down and start from scratch so I can get it right.