fivdi/epoll #21: Documentation Request: Can't install on Non-Linux systems

lyzadanger, 2016-06-28:
Once I found the information that you can't install this module on non-Linux systems (https://github.com/fivdi/epoll/issues/12#issuecomment-138341563) it totally made sense—but I spent about 45 minutes trying to figure out what was going on before I was able to find that thread. The errors one gets suggest problems with `node-gyp` or `bindings`, which are both red herrings.

Would it be possible to add something to the README just calling this out, to save future people a little heartache and time?

fivdi, 2016-06-29:
Sorry about that. You're quite right. I'll add the appropriate documentation.

fivdi, 2016-07-22:
This has now been documented in the [Installation section](https://github.com/fivdi/epoll#installation) of the README.

@lyzadanger thanks for making the suggestion.

closed 2016-07-22 by fivdi