fivdi/epoll #23: Failed at the epoll@0.1.22 install script 'node-gyp rebuild'.

AdamorfuX, 2017-08-26:
I get error message while installing epoll.

Here is error mesage:

npm ERR! epoll@0.1.22 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.22 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs epoll
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls epoll

fivdi, 2017-08-26:
The install works fine for me. Please post the complete command used for the install and the complete screen output of that install command.

AdamorfuX, 2017-08-26:
I tried to reinstall ndm and now it gives me also a error. I dont know what to do.

fivdi, 2017-08-26:
What error? The same error or a different error?

AdamorfuX, 2017-08-26:
Give me a sec i have problems with powering up the pi

AdamorfuX, 2017-08-26:
I gonna reinstall raspian and try again

AdamorfuX, 2017-08-28:
Ok. I fixed it via copying nvm etc to /usr/local/.

fivdi, 2017-08-28:
Ok. Good to hear that it's fixed now.

closed 2017-08-28 by fivdi