fivdi/epoll #25: Timeout waiting for hardware interrupt

adonespitogo, 2018-03-10:
I was able to listen to GPIO input, but after several seconds, I get the ff error:

```
[  687.896153] mmc0: Timeout waiting for hardware interrupt.
[  687.901399] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[  687.908151] mmc0: sdhci: Sys addr:  0x00000000 | Version:  0x00000002
[  687.914725] mmc0: sdhci: Blk size:  0x00007200 | Blk cnt:  0x00000000
[  687.921296] mmc0: sdhci: Argument:  0x018488b8 | Trn mode: 0x00000023
[  687.928229] mmc0: sdhci: Present:   0x01f00000 | Host ctl: 0x00000013
[  687.935071] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000000
[  687.941383] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x00000807
[  687.948313] mmc0: sdhci: Timeout:   0x0000000e | Int stat: 0x00000003
[  687.954977] mmc0: sdhci: Int enab:  0x02ff004b | Sig enab: 0x02ff004b
[  687.961730] mmc0: sdhci: AC12 err:  0x00000000 | Slot int: 0x00000001
[  687.968310] mmc0: sdhci: Caps:      0x25e80099 | Caps_1:   0x0000af77
[  687.974794] mmc0: sdhci: Cmd:       0x0000193a | Max curr: 0x00000000
[  687.981727] mmc0: sdhci: Resp[0]:   0x00000900 | Resp[1]:  0x00000000
[  687.988217] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[  687.994791] mmc0: sdhci: Host ctl2: 0x00000000
[  687.999565] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x3b07e208
[  688.005959] mmc0: sdhci: ============================================
```

Here is my code snippet:
```
function onChange(val) {

        if (has_detected) {
          time_now += interval
          diff = time_now - last_detect_time

          console.log('diff: ', diff)

          if (diff > interval * 3) {
            //console.log('payment: ', current_total)
            payment.received(current_total);
            current_total = 0
            diff = 0
            time_now = 0
            last_detect_time = 0
            has_detected = false
          }
        }

        if (prev === 1 && val === 0) {
          has_detected = true
          last_detect_time = time_now
          current_total += 1
        }

        prev = val

      }

      let valuefd = fs.openSync(gpio_val, 'r')
      let buffer = new Buffer(1)
      let poller = new Epoll((err, fd, events) => {
        // Read GPIO value file. Reading also clears the interrupt.
        fs.readSync(fd, buffer, 0, 1, 0)
        let v = buffer.toString() === '1' ? 1 : 0
        onChange(v)
      })

      // Read the GPIO value file before watching to
      // prevent an initial unauthentic interrupt.
      fs.readSync(valuefd, buffer, 0, 1, 0);

      // Start watching for interrupts.
      poller.add(valuefd, Epoll.EPOLLPRI);
```

fivdi, 2018-03-10:
What type of computer is epoll being run on, a Raspberry Pi?

Which GPIO would you like to detect interrupts on? It looks like it might be a GPIO that has something to do with an SD card reader.

adonespitogo, 2018-03-11:
Im running it on Espressobin with Archlinux ARM kernel `4.15.7-1-ARCH`. The GPIO pin is `GPIO2_5` (pin 11 on the rightmost) 

[image]

adonespitogo, 2018-03-11:
Right now I'm polling the value from `/sys/class/gpio/gpio451/value` at `10ms` interval to get around this.

adonespitogo, 2018-03-11:
These are the gpiochips numbering:
```
[root@portal alarm]# ls /sys/class/gpio
export	gpio451  gpiochip446  gpiochip476  unexport
```

fivdi, 2018-03-11:
I'm not familiar with the Espressobin. You'll need to figure out to what extent it supports interrupts on GPIOs. In this case you'll also need to figure out why processing interrupts on GPIO451 appears to results in issues with mmc0. I'd suggest looking for help at the Espressobin forums. I'm fairly confident that this issues isn't related to the epoll module.

adonespitogo, 2018-03-12:
It's working now. This issue only happens randomly when I insert/unplug a USB flash drive.

closed 2018-03-12 by adonespitogo