fivdi/epoll #27: Build failing on RPI 2

williamluke4, 2018-06-13:
Having a issue install a package ([**rpi-gpio**](https://github.com/JamesBarwell/rpi-gpio.js)) that uses epoll. As far as I can tell it is epolls' build that is failing..  Any pointers?
```bash 
pi@raspberrypi:~/server $ npm install

> epoll@0.1.22 install /home/pi/server/node_modules/epoll
> node-gyp rebuild

gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make: Entering directory '/home/pi/server/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../../nan/nan.h:192:0,
                 from ../src/epoll.cc:15:
../../nan/nan_maybe_43_inl.h: In function ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’:
../../nan/nan_maybe_43_inl.h:112:15: error: ‘class v8::Object’ has no member named ‘ForceSet’
   return obj->ForceSet(isolate->GetCurrentContext(), key, value, attribs);
               ^
In file included from ../src/epoll.cc:15:0:
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’:
../../nan/nan.h:835:60: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated (declared at /home/pi/.node-gyp/10.4.1/include/node/node.h:171): Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, func, argc, argv);
                                                            ^
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’:
../../nan/nan.h:850:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated (declared at /home/pi/.node-gyp/10.4.1/include/node/node.h:164): Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, symbol, argc, argv);
                                                              ^
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’:
../../nan/nan.h:865:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’ is deprecated (declared at /home/pi/.node-gyp/10.4.1/include/node/node.h:157): Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, method, argc, argv);
                                                              ^
../../nan/nan.h: In member function ‘v8::Local<v8::Value> Nan::Callback::Call_(v8::Isolate*, v8::Local<v8::Object>, int, v8::Local<v8::Value>*) const’:
../../nan/nan.h:1479:5: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated (declared at /home/pi/.node-gyp/10.4.1/include/node/node.h:171): Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
     ));
     ^
epoll.target.mk:95: recipe for target 'Release/obj.target/epoll/src/epoll.o' failed
make: *** [Release/obj.target/epoll/src/epoll.o] Error 1
make: Leaving directory '/home/pi/server/node_modules/epoll/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/.nvm/versions/node/v10.4.1/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:237:12)
gyp ERR! System Linux 4.4.13-v7+
gyp ERR! command "/home/pi/.nvm/versions/node/v10.4.1/bin/node" "/home/pi/.nvm/versions/node/v10.4.1/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/server/node_modules/epoll
gyp ERR! node -v v10.4.1
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN server@1.0.0 No description
npm WARN server@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! epoll@0.1.22 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.22 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-06-13T01_53_41_156Z-debug.log
pi@raspberrypi:~/server $ 
```

fivdi, 2018-06-13:
rpi-gpio v1.0.0 depends on epoll v0.1.22 (see [here](https://github.com/JamesBarwell/rpi-gpio.js/blob/v1.0.0/package.json#L36).) epoll v0.1.22 doesn't support Node.js v10.4.1 which is the version of Node.js that you are using. In order to support Node.js v10.4.1 rpi-gpio would need to upgrade it's dependencies and use epoll v2.0.1.

In other words, this is an rpi-gpio issue. Please open an issue as rpi-gpio asking the author to upgrade to upgrade to epoll v2.0.1.

closed 2018-06-13 by fivdi

williamluke4, 2018-06-13:
@fivdi Thanks for the help