fivdi/epoll #30: Can't install epoll module on raspian

gingters, 2018-10-17:
Installing the epoll module on my Raspberry Pi 3 fails with the following error:

```
> epoll@2.0.6 install /home/pi/bsc-ampel/node_modules/epoll
> node-gyp rebuild

Traceback (most recent call last):
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py", line 13, in <module>
    import gyp
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 8, in <module>
    import gyp.input
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 5, in <module>
    from compiler.ast import Const
  File "/usr/lib/python2.7/compiler/__init__.py", line 29, in <module>
    from compiler.transformer import parse, parseFile
  File "/usr/lib/python2.7/compiler/transformer.py", line 28, in <module>
    from compiler.ast import *
ValueError: bad marshal data (unknown type code)
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:345:16)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:238:12)
gyp ERR! System Linux 4.14.52-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/bsc-ampel/node_modules/epoll
gyp ERR! node -v v10.8.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
```

To give some context, I use it in the following project: https://github.com/gingters/bsc-ampel
I did a fresh `npm install` and that fails with the aforementioned problem.
I then have to rebuild that for electron, to use that in the app, but already the first installation and compilation for node fails.

That worked two or three weeks ago (with an older version, I think 2.0.3), and now I don't know how to solve that. Please advise.

fivdi, 2018-10-17:
Please post the complete output of `npm install`.

fivdi, 2018-10-17:
bsc-ampel (which depends on epoll@2.0.3 via `package-lock.json`) installs successfully for me on a Raspberry Pi:
```
pi@raspberrypi:~ $ node -v
v10.8.0
pi@raspberrypi:~ $ git clone https://github.com/gingters/bsc-ampel
Cloning into 'bsc-ampel'...
remote: Enumerating objects: 7, done.
remote: Counting objects: 100% (7/7), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 40 (delta 0), reused 1 (delta 0), pack-reused 33
Unpacking objects: 100% (40/40), done.
pi@raspberrypi:~ $ cd bsc-ampel/
pi@raspberrypi:~/bsc-ampel $ npm install

> epoll@2.0.3 install /home/pi/bsc-ampel/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/bsc-ampel/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/bsc-ampel/node_modules/epoll/build'

> websocket@1.0.26 install /home/pi/bsc-ampel/node_modules/websocket
> (node-gyp rebuild 2> builderror.log) || (exit 0)

make: Entering directory '/home/pi/bsc-ampel/node_modules/websocket/build'
  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node
  COPY Release/bufferutil.node
  CXX(target) Release/obj.target/validation/src/validation.o
  SOLINK_MODULE(target) Release/obj.target/validation.node
  COPY Release/validation.node
make: Leaving directory '/home/pi/bsc-ampel/node_modules/websocket/build'

> electron@2.0.8 postinstall /home/pi/bsc-ampel/node_modules/electron
> node install.js

Downloading SHASUMS256.txt
[============================================>] 100.0% of 5.33 kB (5.33 kB/s)
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 769 packages from 449 contributors and audited 5339 packages in 121.507s
found 0 vulnerabilities

pi@raspberrypi:~/bsc-ampel $ 
```
If `package-lock.json` is removed before calling `npm install` to force epoll@2.0.6 to be installed it also works:
```
pi@raspberrypi:~ $ node -v
v10.8.0
pi@raspberrypi:~ $ git clone https://github.com/gingters/bsc-ampel
Cloning into 'bsc-ampel'...
remote: Enumerating objects: 7, done.
remote: Counting objects: 100% (7/7), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 40 (delta 0), reused 1 (delta 0), pack-reused 33
Unpacking objects: 100% (40/40), done.
pi@raspberrypi:~ $ cd bsc-ampel/
pi@raspberrypi:~/bsc-ampel $ rm package-lock.json 
pi@raspberrypi:~/bsc-ampel $ npm install

> epoll@2.0.6 install /home/pi/bsc-ampel/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/bsc-ampel/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/pi/bsc-ampel/node_modules/epoll/build'

> websocket@1.0.28 install /home/pi/bsc-ampel/node_modules/websocket
> (node-gyp rebuild 2> builderror.log) || (exit 0)

make: Entering directory '/home/pi/bsc-ampel/node_modules/websocket/build'
  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node
  COPY Release/bufferutil.node
  CXX(target) Release/obj.target/validation/src/validation.o
  SOLINK_MODULE(target) Release/obj.target/validation.node
  COPY Release/validation.node
make: Leaving directory '/home/pi/bsc-ampel/node_modules/websocket/build'

> electron@2.0.11 postinstall /home/pi/bsc-ampel/node_modules/electron
> node install.js

Downloading SHASUMS256.txt
[============================================>] 100.0% of 5.39 kB (5.39 kB/s)
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN ajv-keywords@3.2.0 requires a peer of ajv@^6.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 762 packages from 439 contributors and audited 6945 packages in 137.853s
found 0 vulnerabilities

pi@raspberrypi:~/bsc-ampel $ 
```
Everything looks good to me.

gingters, 2018-10-17:
This was pretty much it. I traced the problem down to something with the python installation and did an `apt-get upgrade`, after which a fresh `npm install` went through without problems. Sorry to bother you. I got it sorted out that way.

closed 2018-10-17 by gingters