fivdi/epoll #31: Installing dummy module for Windows

keaton-freude, 2019-01-17:
Hi --

I know this project makes no sense in running on Windows, but I do all of my development on a Windows machine before doing testing on my Raspberry Pi. As such, in my code I've abstracted away the actual calls to rpi-gpio (which uses epoll) such that I never actually invoke any code which hits epoll.

However, Node still complains that the module did not self-register even though the methods will not be called.

A quick work-around I had was to modify epoll.cc and epoll.h slightly to provide an implementation for Windows platform which does nothing. Only registers the module and provides no implementation for the API. I assume I could just no-op all of the methods so it won't blow up when trying to call them, will test this out.

My question is, would such a contribution be useful for this project? Or should I keep this in my own fork (and subsequent forked npm package)?

Thanks for the useful project!

fivdi, 2019-03-01:
@keaton-freude I'm sorry for not responding to this issue earlier.

I'm going to attempt to resolve this issue and the related issues that @grocky and @iammatthew2 are having on macOS/OSX [here](https://github.com/fivdi/onoff/pull/122) and [here](https://github.com/fivdi/epoll/issues/31).

keaton-freude, 2019-03-01:
Cool. It's no problem. I did it in my project and thought I'd offer.

Keaton

fivdi, 2019-03-02:
@keaton-freude epoll v2.0.8 has been published on npm and although I haven't tried it out on Windows, I think you should be able to achieve what you want to achieve now. That being said, as you are using rpi-gpio, rpi-gpio may need to be updated to use epoll v2.0.8 too.

It would be great if you could give epoll v2.0.8 a try to see it it works as I don't have access to a Windows computer to try it out.

keaton-freude, 2019-03-02:
Will do. I'll try and do that and update today.

fivdi, 2019-03-06:
I finally managed to get my hands on a Windows computer with all the required software installed. The good news is that epoll installs successfully on Windows:
```
C:\src\epoll>npm i epoll

> epoll@2.0.8 install C:\src\epoll\node_modules\epoll
> node-gyp rebuild

C:\src\epoll\node_modules\epoll>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  win_delay_load_hook.cc
  Generating code
  All 1 functions were compiled because no usable IPDB/IOBJ from previous compilation was found.
  Finished generating code
  epoll.vcxproj -> C:\src\epoll\node_modules\epoll\build\Release\\epoll.node
  epoll.vcxproj -> C:\src\epoll\node_modules\epoll\build\Release\epoll.pdb (Full PDB)
npm WARN saveError ENOENT: no such file or directory, open 'C:\src\epoll\package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open 'C:\src\epoll\package.json'
npm WARN epoll No description
npm WARN epoll No repository field.
npm WARN epoll No README data
npm WARN epoll No license field.

+ epoll@2.0.8
added 4 packages from 9 contributors and audited 4 packages in 3.813s
found 0 vulnerabilities

C:\src\epoll>
```
It's also now possible to "require('epoll')" without geting module did not self-register errors:
```
C:\src\epoll>node -v
v10.15.3

C:\src\epoll>node
> const Epoll = require('epoll').Epoll;
Warning: epoll is built for Linux and not intended for usage on Windows_NT.
undefined
> Epoll
{}
>
```

closed 2019-03-06 by fivdi