fivdi/epoll #34: Add prebuilt binaries?

flotwig, 2019-03-17:
It would be nice if the `epoll` library had some prebuilt binaries shipping with it, either using the low-level [`node-pre-gyp`](https://github.com/mapbox/node-pre-gyp) or the higher-level [`prebuild`](https://github.com/prebuild/prebuild) and [`prebuild-install`](https://github.com/prebuild/prebuild-install) libraries. If `epoll` came pre-built for the Raspberry Pi architecture, users wouldn't have to install Python, make, or g++ to install the package anymore, which would make it a lot faster for new users to use.

fivdi, 2019-03-17:
> It would be nice if the epoll library had some prebuilt binaries shipping with it, either using the low-level node-pre-gyp or the higher-level prebuild and prebuild-install libraries.

Are there any projects out there that use prebuilt binaries for any of the ARM architectures? For example, for the ARMv6 architecture of the Raspberry Pi Zero and 1, or the ARMv7 architecture of the Raspberry Pi 2 or or the ARMv8 architecture of the Raspberry Pi 3? To the best of my knowledge there are no projects using prebuilt binaries for any of the ARM architectures. 

> If epoll came pre-built for the Raspberry Pi architecture, users wouldn't have to install Python, make, or g++ to install the package anymore, which would make it a lot faster for new users to use.

Hmm... What distribution of Linux that is suitable for beginners doesn't come with all of these tools pre-installed?

flotwig, 2019-03-17:
> Are there any projects out there that use prebuilt binaries for any of the ARM architectures? For example, for the ARMv6 architecture of the Raspberry Pi Zero and 1, or the ARMv7 architecture of the [...]

I'm not sure, but it is possible.

> Hmm... What distribution of Linux that is suitable for beginners doesn't come with all of these tools pre-installed?

DietPi doesn't, can't speak for others.

fivdi, 2019-03-17:
> Are there any projects out there that use prebuilt binaries for any of the ARM architectures? For example, for the ARMv6 architecture of the Raspberry Pi Zero and 1, or the ARMv7 architecture of the [...]

>I'm not sure, but it is possible.

I'm going to make the claim that it's not possible until I see evidence of it being done in practice. I have **never** seen prebuilt binaries for any of the ARM architectures. I may be reading between the line here but it also sounds like you have never seen any either :smile:

>>Hmm... What distribution of Linux that is suitable for beginners doesn't come with all of these tools pre-installed?

>DietPi doesn't, can't speak for others.

DietPi is an extremely lightweight distribution and as such it's definitely not suitable for beginners. It's only suitable for experts. On the Raspberry Pi, Raspbian is suitable for beginners and comes with all the required tools pre-installed.

> btw - thanks for your work on the onoff library, it's awesome to be able to use GPIO in node.js!

Thank you very much.

Creating prebuilt binaries for the ARM architectures, if even possible, would be a non-trivial, challenging and time consuming task. It would also require infrastructure for testing all the prebuilt binaries to ensure that they actually work. It's not something that I want to work on so I'm going to close this as a `wontfix`.

closed 2019-03-17 by fivdi

flotwig, 2019-03-17:
Well, if you do want to do it at some point in the future, `node-pre-gyp` supports pre-building for ARM, you just need to specify `--target_arch=arm` :)

https://github.com/mapbox/node-pre-gyp#options

fivdi, 2019-03-17:
> you just need to specify --target_arch=arm :)

And this is where the problems start. Which of the several ARM architectures is being referred to here? ARMv6? ARMv7, ARMv8? How is 32bit specified? How is 64bit specified?

flotwig, 2019-03-17:
ARMv7 and ARMv6 are both 32-bit, and you can built them by using the `armv7` and `armv6` target archs I believe - `node-gyp` uses the architecture names you'd expect. ARMv8 is `arm`, and ARMv8 64-bit is `arm64`.

fivdi, 2019-03-17:
>you can built them by using the armv7 and armv6 target archs I believe

According to open issue https://github.com/mapbox/node-pre-gyp/issues/348 this doesn't work.

If you can provide a link to a project that provides prebuilt binaries for all three ARM architectures of interest we can discuss the topic.

flotwig, 2019-03-17:
Damn, that sucks. If only one of us were motivated enough to submit a PR and fix that issue!

I guess this will have to wait until mapbox/node-pre-gyp#348 is fixed.