fivdi/epoll #35: npm install onoff on raspberry PI 512 MB DE

agrillea1234, 2019-04-27:
root@raspberrypi:~# **npm install onoff**

> epoll@2.0.9 install /root/node_modules/epoll
> node-gyp rebuild

make: ingresso nella directory "/root/node_modules/epoll/build"
  CXX(target) Release/obj.target/epoll/src/epoll.o
make: g++: comando non trovato
epoll.target.mk:96: set di istruzioni per l'obiettivo "Release/obj.target/epoll/src/epoll.o" non riuscito
make: *** [Release/obj.target/epoll/src/epoll.o] Errore 127
make: uscita dalla directory "/root/node_modules/epoll/build"
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.1.19+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/node_modules/epoll
gyp ERR! node -v v6.11.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/root/package.json'
npm WARN root No description
npm WARN root No repository field.
npm WARN root No README data
npm WARN root No license field.
npm ERR! Linux 4.1.19+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "onoff"
npm ERR! node v6.11.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! epoll@2.0.9 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the epoll@2.0.9 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs epoll
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/npm-debug.log
root@raspberrypi:~# npm bugs epoll
npm ERR! Linux 4.1.19+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "bugs" "epoll"
npm ERR! node v6.11.1
npm ERR! npm  v3.10.10
npm ERR! code 3

npm ERR! Command failed: xdg-open https://github.com/fivdi/epoll/issues
npm ERR! /usr/bin/xdg-open: 1: eval: www-browser: not found
npm ERR! /usr/bin/xdg-open: 1: eval: links2: not found
npm ERR! /usr/bin/xdg-open: 1: eval: elinks: not found
npm ERR! /usr/bin/xdg-open: 1: eval: links: not found
npm ERR! /usr/bin/xdg-open: 1: eval: lynx: not found
npm ERR! /usr/bin/xdg-open: 1: eval: w3m: not found
npm ERR! xdg-open: no method available for opening 'https://github.com/fivdi/epoll/issues'
npm ERR!
npm ERR!
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /root/npm-debug.log
root@raspberrypi:~# nano /root/npm-debug.log
  GNU nano 2.2.6                             File: /root/npm-debug.log

fivdi, 2019-04-27:
It's probably not a good idea to login as user `root` to install npm packages like onoff. If this must be done it will probably only work with the following command:
```
npm install --unsafe-perm onoff
```
If you are using Raspbian, why not login as user `pi` and install with command `npm install onoff`?

agrillea1234, 2019-04-27:
Thank’s I  resolve with this snippet of command:

sudo apt-get install g++-4.7
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-4.6 60
--slave /usr/bin/g++ g++ /usr/bin/g++-4.6
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-4.7 40
--slave /usr/bin/g++ g++ /usr/bin/g++-4.7
sudo update-alternatives --config gcc

select  g++4.7

npm install onoff has worked.

by

Inviato da Posta per Windows 10

Da: Brian Cooke
Inviato: sabato 27 aprile 2019 22:09
A: fivdi/epoll
Cc: agrillea1234; Author
Oggetto: Re: [fivdi/epoll] npm install onoff on raspberry PI 512 MB DE
(#35)

fivdi, 2019-04-28:
Good to gear that it's working now. It looks like the version of g++ that was installed was too old. Thank you for providing the feedback.

closed 2019-04-28 by fivdi

Snowworm3000, 2020-06-14:
> It's probably not a good idea to login as user `root` to install npm packages like onoff. If this must be done it will probably only work with the following command:
>
> ```
> npm install --unsafe-perm [...]

This was very helpfull and it solved my problem