fivdi/epoll #36: Ubuntu 18.04 Server fails to install @node-gyp rebuild

engineertdog, 2019-07-05:
I'm running a clean install of Ubuntu 18.04 Server on a raspberry PI with Node v10.16.0 and NPM v6.9.0. As the ubuntu user, I am trying to install onoff, and only onoff, with one npm install command. Attached is the full output from `npm install onoff`

> 0 info it worked if it ends with ok
> 1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'install', 'onoff' ]
> 2 info using npm@6.9.0
> 3 info using node@v10.16.0
> 4 verbose npm-session 16df9898dbc21df5
> 5 silly install loadCurrentTree
> 6 silly install readLocalPackageData
> 7 http fetch GET 200 https://registry.npmjs.org/onoff 103ms (from cache)
> 8 silly pacote tag manifest for onoff@latest fetched in 155ms
> 9 timing stage:loadCurrentTree Completed in 1108ms
> 10 silly install loadIdealTree
> 11 silly install cloneCurrentTreeToIdealTree
> 12 timing stage:loadIdealTree:cloneCurrentTree Completed in 9ms
> 13 silly install loadShrinkwrap
> 14 timing stage:loadIdealTree:loadShrinkwrap Completed in 513ms
> 15 silly install loadAllDepsIntoIdealTree
> 16 silly resolveWithNewModule onoff@4.1.2 checking installable status
> 17 http fetch GET 200 https://registry.npmjs.org/epoll 176ms (from cache)
> 18 silly pacote range manifest for epoll@^2.0.9 fetched in 197ms
> 19 silly resolveWithNewModule epoll@2.0.9 checking installable status
> 20 http fetch GET 304 https://registry.npmjs.org/lodash.debounce 421ms (from cache)
> 21 silly pacote range manifest for lodash.debounce@^4.0.8 fetched in 431ms
> 22 silly resolveWithNewModule lodash.debounce@4.0.8 checking installable status
> 23 http fetch GET 200 https://registry.npmjs.org/bindings 54ms (from cache)
> 24 silly pacote range manifest for bindings@^1.5.0 fetched in 63ms
> 25 silly resolveWithNewModule bindings@1.5.0 checking installable status
> 26 http fetch GET 304 https://registry.npmjs.org/nan 123ms (from cache)
> 27 silly pacote range manifest for nan@^2.13.0 fetched in 135ms
> 28 silly resolveWithNewModule nan@2.14.0 checking installable status
> 29 http fetch GET 304 https://registry.npmjs.org/file-uri-to-path 84ms (from cache)
> 30 silly pacote version manifest for file-uri-to-path@1.0.0 fetched in 91ms
> 31 silly resolveWithNewModule file-uri-to-path@1.0.0 checking installable status
> 32 timing stage:loadIdealTree:loadAllDepsIntoIdealTree Completed in 1353ms
> 33 timing stage:loadIdealTree Completed in 2041ms
> 34 silly currentTree pilets@1.0.0
> 34 silly currentTree ├── accepts@1.3.7
> 34 silly currentTree ├── array-flatten@1.1.1
> 34 silly currentTree ├── body-parser@1.19.0
> 34 silly currentTree ├── bytes@3.1.0
> 34 silly currentTree ├── content-disposition@0.5.3
> 34 silly currentTree ├── content-type@1.0.4
> 34 silly currentTree ├── cookie-signature@1.0.6
> 34 silly currentTree ├── cookie@0.4.0
> 34 silly currentTree ├── cors@2.8.5
> 34 silly currentTree ├── debug@2.6.9
> 34 silly currentTree ├── depd@1.1.2
> 34 silly currentTree ├── destroy@1.0.4
> 34 silly currentTree ├── ee-first@1.1.1
> 34 silly currentTree ├── encodeurl@1.0.2
> 34 silly currentTree ├── escape-html@1.0.3
> 34 silly currentTree ├── etag@1.8.1
> 34 silly currentTree ├── express@4.17.1
> 34 silly currentTree ├── finalhandler@1.1.2
> 34 silly currentTree ├── forwarded@0.1.2
> 34 silly currentTree ├── fresh@0.5.2
> 34 silly currentTree ├── http-errors@1.7.2
> 34 silly currentTree ├── iconv-lite@0.4.24
> 34 silly currentTree ├── inherits@2.0.3
> 34 silly currentTree ├── ipaddr.js@1.9.0
> 34 silly currentTree ├── lodash.debounce@4.0.8
> 34 silly currentTree ├── media-typer@0.3.0
> 34 silly currentTree ├── merge-descriptors@1.0.1
> 34 silly currentTree ├── methods@1.1.2
> 34 silly currentTree ├── mime-db@1.40.0
> 34 silly currentTree ├── mime-types@2.1.24
> 34 silly currentTree ├── mime@1.6.0
> 34 silly currentTree ├── ms@2.0.0
> 34 silly currentTree ├── negotiator@0.6.2
[... 278 lines cut ...]
> 176 silly saveTree   │ └── nan@2.14.0
> 176 silly saveTree   └── lodash.debounce@4.0.8
> 177 warn pilets@1.0.0 No description
> 178 warn pilets@1.0.0 No repository field.
> 179 verbose stack Error: epoll@2.0.9 install: `node-gyp rebuild`
> 179 verbose stack Exit status 1
> 179 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
> 179 verbose stack     at EventEmitter.emit (events.js:198:13)
> 179 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
> 179 verbose stack     at ChildProcess.emit (events.js:198:13)
> 179 verbose stack     at maybeClose (internal/child_process.js:982:16)
> 179 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
> 180 verbose pkgid epoll@2.0.9
> 181 verbose cwd /home/ubuntu/pilets
> 182 verbose Linux 4.15.0-1040-raspi2
> 183 verbose argv "/usr/bin/node" "/usr/bin/npm" "install" "onoff"
> 184 verbose node v10.16.0
> 185 verbose npm  v6.9.0
> 186 error code ELIFECYCLE
> 187 error errno 1
> 188 error epoll@2.0.9 install: `node-gyp rebuild`
> 188 error Exit status 1
> 189 error Failed at the epoll@2.0.9 install script.
> 189 error This is probably not a problem with npm. There is likely additional logging output above.
> 190 verbose exit [ 1, true ]

fivdi, 2019-07-05:
That looks like the contents of the npm log file. Unfortunately this file doesn't include the information printed on the screen when `npm install onoff` in called. Can you post the information printed to the screen when `npm install onoff` is called please.

engineertdog, 2019-07-05:
> ubuntu@ubuntu:~/pilets$ npm install onoff
> 
> > epoll@2.0.9 install /home/ubuntu/pilets/node_modules/epoll
> > node-gyp rebuild
> 
> gyp ERR! build error
> gyp ERR! stack Error: not found: make
> gyp ERR! stack     at getNotFoundError (/usr/lib/node_modules/npm/node_modules/w                                                                                           hich/which.js:13:12)
> gyp ERR! stack     at F (/usr/lib/node_modules/npm/node_modules/which/which.js:6                                                                                           8:19)
> gyp ERR! stack     at E (/usr/lib/node_modules/npm/node_modules/which/which.js:8                                                                                           0:29)
> gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:89:1                                                                                           6
> gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/index.js:42:5
> gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
> gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:153:21)
> gyp ERR! System Linux 4.15.0-1040-raspi2
> gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gy                                                                                           p/bin/node-gyp.js" "rebuild"
> gyp ERR! cwd /home/ubuntu/pilets/node_modules/epoll
> gyp ERR! node -v v10.16.0
> gyp ERR! node-gyp -v v3.8.0
> gyp ERR! not ok
> npm WARN pilets@1.0.0 No description
> npm WARN pilets@1.0.0 No repository field.
> 
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! epoll@2.0.9 install: `node-gyp rebuild`
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the epoll@2.0.9 install script.
> npm ERR! This is probably not a problem with npm. There is likely additional log                                                                                           ging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/ubuntu/.npm/_logs/2019-07-05T15_58_46_668Z-debug.log

fivdi, 2019-07-05:
It looks like the tools that are needed for building C/C++ code have not been installed yet. Run the following command and then try to install onoff again:
```
sudo apt-get install build-essential
```

engineertdog, 2019-07-05:
I'm sorry, I had just noticed that I somehow came to this repo instead of the official onoff repo itself, even though I had used the npm page. That did the trick, but I'm curious as to why this was not previously installed? I could see that being an issue for a fresh Ubuntu 18.04 Server, where it's a little more lightweight for the PI. However, even my full Ubuntu 18.04 AWS EC2 instance could not install onoff either, and I had to install `build-essential` as well.

fivdi, 2019-07-05:
>I'm sorry, I had just noticed that I somehow came to this repo instead of the official onoff repo itself, even though I had used the npm page.

No problem.

>That did the trick

That's good news.

> but I'm curious as to why this was not previously installed? I could see that being an issue for a fresh Ubuntu 18.04 Server, where it's a little more lightweight for the PI. However, even my full [...]

The people who create the Ubuntu distributions probably decided that the majority of their users don't need `build-essentials` so they didn't include it by default. This sounds reasonable to me. Typically `build-essentials` is only needed by users who build software from source. I guess that the majority of Ubuntu users don't do this.

closed 2019-07-05 by fivdi

engineertdog, 2019-07-05:
Would it be recommended to include it in the readme / installation somewhere? I don't recall if I had to do this the last time I installed onoff. But, that was also on raspbian and not on Ubuntu.

fivdi, 2019-07-05:
>Would it be recommended to include it in the readme / installation somewhere? I don't recall if I had to do this the last time I installed onoff. But, that was also on raspbian and not on Ubuntu.

I'd prefer not to add information to the readme that the majority of users won't need. Raspbian is the operating system used by most people on the Raspberry Pi. Raspbian comes with `build-essentials` installed by default. BeagleBobe.org provides Debian distributions for the BeagleBone. This also comes with `build essentials` installed by default. The manufacturers of SBCs typically have developers in focus and will include `build-essentials` with their distributions by default.

If the error occurred on Raspbian I would add it to the readme.