fivdi/epoll #38: epoll 64bit support for Raspberry Pi 4B error epoll.node: wrong ELF class: ELFCLASS64

bmwertman, 2019-11-01:
After giving up on fighting with getting my program running on my Banana Pi M64 I learned that Raspberry Pi had released a V4 with better performance specs, cheaper price than the BPi and more readily available via Amazon Prime. It's been much smoother sailing until I ran into this error.

Does epoll not support 64 bit system architecture? Is there a work-around I can try?

```
Error: /node_modules/epoll/build/Release/epoll.node: wrong ELF class: ELFCLASS64
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1003:18)
    at Module.load (internal/modules/cjs/loader.js:812:32)
    at Function.Module._load (internal/modules/cjs/loader.js:724:14)
    at Module.require (internal/modules/cjs/loader.js:849:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at bindings (node_modules/bindings/bindings.js:112:48)
    at node_modules/epoll/epoll.js:7:31
    at Object.<anonymous> (node_modules/epoll/epoll.js:15:3)
    at Module._compile (internal/modules/cjs/loader.js:956:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:973:10)
```

bmwertman, 2019-11-01:
Removing `node_modules` and re-running `npm install` seemed to fix the issue.

closed 2019-11-01 by bmwertman

fivdi, 2020-09-25:
> Does epoll not support 64 bit system architecture?

Yes, it does.