fivdi/epoll #42: Electron - Requiring Native Modules in the Renderer Process to be NAPI or Context Aware

David-University, 2021-01-15:
As mentioned in the following issue for electron, support for native modules that are not NAPI or Context Aware are going to be dropped.  [Electron Issue](https://github.com/electron/electron/issues/18397)

This will result in applications that use epoll/onoff being broken after Electron 12.

Applications based on electron 10+ require the flag app.allowRendererProcessReuse = false in the Electron main.js.

fivdi, 2021-01-15:
I guess it would make sense to migrate from NAN to N-API to resolve this issue.

@David-University do you use Electron on a Raspberry Pi (or similar)? I don't have any idea of how many people do this.

David-University, 2021-01-16:
I use Electron on a raspberry Pi 3 and raspberry Pi 4.

fivdi, 2021-01-17:
Ok. I'm not sure if I'll get around to this any time soon.

fivdi, 2021-10-29:
Not at the moment.