fivdi/epoll #44: node-gyp installation fails on raspbian

doublemcz, 2021-11-18:
Hi,

I am having struggle with epoll installation on raspberry pi 2 (model B). 

Versions:  

- Raspbian GNU/Linux 11 (bullseye)
- NodeJs 16.13.0
- NPM 8.1.0
- Python 3.9.2

Log from npm install

```
npm ERR! path /home/pi/Projects/rizeni/node_modules/epoll
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@3.8.0
npm ERR! gyp info using node@16.13.0 | linux | arm
npm ERR! gyp ERR! configure error
npm ERR! gyp ERR! stack Error: Command failed: /usr/bin/python -c import sys; print "%s.%s.%s" % sys.version_info[:3];
npm ERR! gyp ERR! stack   File "<string>", line 1
npm ERR! gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
npm ERR! gyp ERR! stack                       ^
npm ERR! gyp ERR! stack SyntaxError: invalid syntax
npm ERR! gyp ERR! stack
npm ERR! gyp ERR! stack     at ChildProcess.exithandler (node:child_process:397:12)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:390:28)
npm ERR! gyp ERR! stack     at maybeClose (node:internal/child_process:1064:16)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
npm ERR! gyp ERR! System Linux 5.10.63-v7+
npm ERR! gyp ERR! command "/usr/bin/node" "/home/pi/Projects/rizeni/node_modules/.bin/node-gyp" "rebuild"
npm ERR! gyp ERR! cwd /home/pi/Projects/rizeni/node_modules/epoll
npm ERR! gyp ERR! node -v v16.13.0
npm ERR! gyp ERR! node-gyp -v v3.8.0
npm ERR! gyp ERR! not ok
```

I can see from the log that it has an issue with getting python version (it's syntax). When I try to do the same command directly in python shell, the error is the same. 

Any hints how to get around this? Do I need to downgrade Python? (I have the latest one at this moment)

Thanks,
Martin

fivdi, 2021-11-18:
I can't reproduce the error. Here is what I see when I install epoll on a Raspberry Pi 4 running Debian Bullseye:

```
pi@raspberrypi:~/test $ cat /etc/debian_version 
11.1
pi@raspberrypi:~/test $ node --version
v16.13.0
pi@raspberrypi:~/test $ npm --version
8.1.0
pi@raspberrypi:~/test $ python --version
Python 2.7.18
pi@raspberrypi:~/test $ python3 --version
Python 3.9.2
pi@raspberrypi:~/test $ npm i epoll

added 4 packages, and audited 5 packages in 7s

found 0 vulnerabilities
pi@raspberrypi:~/test $ 
```

Your system appears to be configured in an unusual way, for example, in the error message shown above, the following can be seen:

> npm ERR! gyp info using node-gyp@3.8.0

node-gyp@3.8.0 is very old and I'm not sure how it got installed on your system or why it's there. As can be seen [here](https://github.com/npm/cli/blob/v8.1.0/package.json#L99), npm@8.1.0 depends on node-gyp@8.2.0.

doublemcz, 2021-11-18:
Thank you for the quick response. 

I can see that problem is with Python 2 vs Python 3. When I install new node-gyp@8.4.0 it screams about versions mismatch (Python2 is not in path etc). And by your commands in the shell, I can see that your default Python version is 2. (Mine was 3 - obvisouly default installation of Raspbian don't offer Python 2 anymore by default). I installed Python 2 by command 

`sudo apt-get install python-packagename`

It wiped python 3 and installed 2. Now the `npm install` works.

Thank you a lot,
Closing...

closed 2021-11-18 by doublemcz

fivdi, 2021-11-18:
Good to hear that it's working now.

> obvisouly default installation of Raspbian don't offer Python 2 anymore by default

I'm using a default installation of Raspberry Pi OS (2021-10-30-raspios-bullseye-armhf-full) which comes with Python 2 and Python 3 pre-installed.