fivdi/epoll #45: rror: static assertion failed: Pointer compression can be enabled only for 64-bit architectures

qistoph, 2022-01-20:
Building with `election-rebuild@3.2.7` fails with:
**error: static assertion failed: Pointer compression can be enabled only for 64-bit architectures**

Any suggestions on how to fix this?

Below is the full log.

```
⠋ Building module: epoll, Completed: 0In file included from /home/pi/.electron-gyp/16.0.5/include/node/v8-local-handle.h:12,
                 from /home/pi/.electron-gyp/16.0.5/include/node/v8-array-buffer.h:12,
                 from /home/pi/.electron-gyp/16.0.5/include/node/v8.h:25,
                 from ../src/epoll.cc:11:
/home/pi/.electron-gyp/16.0.5/include/node/v8-internal.h:113:27: error: static assertion failed: Pointer compression can be enabled only for 64-bit architectures
     kApiSystemPointerSize == kApiInt64Size,
     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
/home/pi/.electron-gyp/16.0.5/include/node/v8-internal.h:471:71: warning: left shift count >= width of type [-Wshift-count-overflow]
   static constexpr size_t kPtrComprCageReservationSize = size_t{1} << 32;
                                                                       ^~
/home/pi/.electron-gyp/16.0.5/include/node/v8-internal.h:471:68: error: right operand of shift expression ‘(1 << 32)’ is >= than the precision of the left operand [-fpermissive]
   static constexpr size_t kPtrComprCageReservationSize = size_t{1} << 32;
                                                          ~~~~~~~~~~^~~~~
/home/pi/.electron-gyp/16.0.5/include/node/v8-internal.h:472:69: warning: left shift count >= width of type [-Wshift-count-overflow]
   static constexpr size_t kPtrComprCageBaseAlignment = size_t{1} << 32;
                                                                     ^~
/home/pi/.electron-gyp/16.0.5/include/node/v8-internal.h:472:66: error: right operand of shift expression ‘(1 << 32)’ is >= than the precision of the left operand [-fpermissive]
   static constexpr size_t kPtrComprCageBaseAlignment = size_t{1} << 32;
                                                        ~~~~~~~~~~^~~~~
⠧ Building module: epoll, Completed: 0make: *** [epoll.target.mk:125: Release/obj.target/epoll/src/epoll.o] Error 1
make: Leaving directory '/home/pi/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll/build'
✖ Rebuild Failed
```

qistoph, 2022-01-20:
Running `npm install` in the epoll dir works correctly:

```
~/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll $ npm install
~/MagicMirror/modules/MMM-PIR-Sensor/node_modules/epoll $ grep -r compression
build/config.gypi:    "v8_enable_pointer_compression": 0,
```

However it seems that using `electron-rebuild` breaks the pointer compression:

```
~/MagicMirror/modules/MMM-PIR-Sensor $ node_modules/.bin/electron-rebuild -e ../../node_modules/electron
[...]
/home/pi/.electron-gyp/16.0.5/include/node/v8-internal.h:113:27: error: static assertion failed: Pointer compression can be enabled only for 64-bit architectures
     kApiSystemPointerSize == kApiInt64Size,
[...]
~/MagicMirror/modules/MMM-PIR-Sensor $ grep -r compression node_modules/epoll/
node_modules/epoll/build/config.gypi:    "v8_enable_pointer_compression": 1,
```

fivdi, 2022-01-20:
What version of Electron are you using?

I haven't used Electron myself but to the best of my knowledge, native modules such as epoll that are not "Context Aware" will not work with Electrom 14 or higher. See https://github.com/electron/electron/issues/18397.

qistoph, 2022-01-21:
```
~/MagicMirror/modules/MMM-PIR-Sensor $ DEBUG="electron-rebuild" ./node_modules/.bin/electron-rebuild -e ../../node_modules/electron
⠋ Searching dependency tree  electron-rebuild rebuilding with args: [Arguments] {
  '0': {
    buildPath: '/home/pi/MagicMirror/modules/MMM-PIR-Sensor',
    electronVersion: '16.0.5',
    arch: 'arm',
    extraModules: [],
    onlyModules: null,
    force: undefined,
    headerURL: undefined,
    types: [ 'prod', 'optional' ],
    mode: undefined,
    debug: undefined,
    prebuildTagPrefix: 'v',
    forceABI: undefined,
    useElectronClang: false,
    disablePreGypCopy: false,
    projectRootPath: '/home/pi/MagicMirror/modules/MMM-PIR-Sensor'
  }
} +0ms
```

I managed to find a work around, but don't understand enough of gyp to comprehend a proper fix.

I've found that `~/.electron-gyp/16.0.5/include/node/common.gypi` contains rules to set `v8_enable_pointer_compression: 1` only if `target_arch` is `arm64` or `x64`. Which shouldn't be the case, since I'm running this on arm.

Setting `v8_enable_pointer_compression: 0` in `node_modules/node-gyp/addon.gypi`'s `variables` does get my electron-rebuild run.